sharpie7/circuitjs1 #768: GPU process isn't usable. Goodbye.

mabret, 2022-05-10:
When I try to run circuitjs1, a window appears briefly then disappears.  The program says:
```
8057:0509/223206.357874:FATAL:gpu_data_manager_impl_private.cc(439)] GPU process isn't usable. Goodbye.
Trace/breakpoint trap (core dumped)
```
It exits with code 133.

`inxi` gives the following output:
```
CPU: quad core Intel Core i7-3770 (-MT MCP-)
speed/min/max: 1741/1600/3900 MHz Kernel: 5.15.0-27-lowlatency x86_64 Up: 42m
Mem: 2382.3/32060.7 MiB (7.4%) Storage: 2.05 TiB (23.1% used) Procs: 287
Shell: Bash inxi: 3.3.13
```
Nvidia GT1030, driver version 470.103.01

ormaaj, 2022-08-13:
What leads you to think this is related to circuitjs? Did you build this to run under electron? This problem could originate almost anywhere _but_ circuitjs - the browser, any library linked against it, the nvidia driver, any part of the graphics stack - we can't know without a backtrace, system logs, and possibly other tracing depending on where the issue is.

> It exits with code 133

The max exit status is 127.

`bash(1)` says:
> The return value of a simple command is its exit status, or 128+n if the command is terminated by signal n.

Use `kill -l` to print the signal list.

`signal(7)` says:
> There are six signals that can be delivered as a consequence of a hardware exception: SIGBUS, SIGEMT, SIGFPE, SIGILL, SIGSEGV, and SIGTRAP.  Which of these signals is delivered, for any given hardware exception, is not documented and  does  not  always  make sense.

That's you're only clue. Your video card could be dying.

pfalstad, 2022-08-14:
It seems to be an electron bug which has since been fixed.  You could try using --no-sandbox or --disable-gpu-sandbox, or download a newer version of electron and run it in the same directory as circuitjs1 (circuitjs1 is just an electron binary that's been renamed).  I don't have linux so I can't try it.

thiago1255, 2023-07-01:
> It seems to be an electron bug which has since been fixed. You could try using --no-sandbox or --disable-gpu-sandbox, or download a newer version of electron and run it in the same directory as [...]

I had the same problem, run with "--disable-gpu-sandbox" works ! thx

blackPantherOS, 2025-05-24:
Nvidia GTX 4070 (550.144 - 575.51.x drivers)

```

[595446:0524/074835.998198:FATAL:gpu_data_manager_impl_private.cc(439)] GPU process isn't usable. Goodbye.
Trace/breakpoint trap (core dumped)
[595637:0524/074837.186691:FATAL:gpu_data_manager_impl_private.cc(439)] GPU process isn't usable. Goodbye.
Trace/breakpoint trap (core dumped)
[595766:0524/074838.247960:FATAL:gpu_data_manager_impl_private.cc(439)] GPU process isn't usable. Goodbye.
Trace/breakpoint trap (core dumped)
[595890:0524/074839.289493:FATAL:gpu_data_manager_impl_private.cc(439)] GPU process isn't usable. Goodbye.
Trace/breakpoint trap (core dumped)
[596013:0524/074840.237864:FATAL:gpu_data_manager_impl_private.cc(439)] GPU process isn't usable. Goodbye.
Trace/breakpoint trap (core dumped)
[596137:0524/074841.154977:FATAL:gpu_data_manager_impl_private.cc(439)] GPU process isn't usable. Goodbye.

```

This bug still exists in the used Electron. Please upgrade to latest Electron version.