sharpie7/circuitjs1 #769: Forward Subcircuits

TimFFischer, 2022-05-18:
Is there an easy way to forward the interieur of multiple subcircuits, which are used in another circuit to somebody else? 
I would like to share working onto one circuit with more than a dozen subcircuits. Currently I share each subcircuit as text and the new colleagues insert these subsequently.

TimFFischer, 2022-05-19:
Edit: Is there maybe a way for the standalone version to forward circuits including subcircuits easier? By this, I would ask the students to use his version for developing. We currently try to make logic circuits for most of the AVR peripherals with registers as introductional examples (SPI, I2C, ADC, T/C etc. )

pfalstad, 2022-05-20:
So when saving a circuit, you want to save the implementation (the subcircuit layout) so they can be viewed?

Or maybe, when creating a subcircuit, I could add an option to force the implementation to be saved.  Because we wouldn't have the implementation available otherwise.

ormaaj, 2022-08-17:
I would try to find a decent library implementation for [pax](https://pubs.opengroup.org/onlinepubs/9699919799/utilities/pax.html#tag_20_92_13_01) to just save and load projects with their associated subcircuits and models as a tarball. Embedding them through escaped strings is going to be pretty nasty especially when nesting. Also way easier for users to untar the file to get at the contents than to figure out decoding the escaping scheme and write a custom tool for parsing the monolithic save format.

pfalstad, 2022-08-17:
I'd like to keep it a text file if possible.  The escaped strings are ugly but we can get rid of those without completely changing the file format.

ormaaj, 2022-08-18:
you could do both.  Not really a total change, just a way to bundle multiple files for saving and loading a collection of them at once. I guess there are other ways, an archive is just the easiest way that doesn't involve completely changing the format.

I think every option requires at least some change because you need a way to reference the subcircuits. That isn't a huge deal I think. You can always create a conversion tool to convert old save files to a new format.

Rather than referencing other files in an archive you could encode other files as a base64 blob and embed the blob within the single save file, and add a way to refer to it by reference. That's still pretty easy to figure out decoding using standard tools and keeps the save format a text-only single file if you prefer that.

fritzw, 2022-11-29:
This would be a huge improvement, especially for teaching the function and design of electronic circuits. The current workflow editing subcircuits feels very awkward (in fact I have not yet understood it completely), especially when compared to the workflow on https://circuitverse.org/simulator for example, where you can edit all sub-circuits at any time by switching tabs in the simulator.

I'm considering the use of this simulator for the practice assignments in our university course, but the current sub-circuit functionality makes me skeptical how well it would be adopted by our students. Currently we offer some practice circuits on circuitverse.org, which use subcircuits to implement counters for example (see https://circuitverse.org/users/132113). If you implemented some kind of sub-circuit editing capability, with the sub-circuits embedded in the saved text file, that would make the decision much easier. Of course, this feature would need a good intuitive workflow.

Let me know if I can help in any way (programming, concepts, or otherwise).