sharpie7/circuitjs1 #771: Feature- small component size also for subcircuits  (in small grid)

TimFFischer, 2022-05-22:
It would be great to be have the small scale components also available for the subcircuits.

pfalstad, 2022-05-22:
I forgot about small grid.  That feature was more useful before zoom was added.  Now you don't need it as much anymore.

TimFFischer, 2022-05-22:
The not scalable texts, like label text or logic IO state would be better readable then the other components are in small scale.

An example for this is [my first try to simulate AVRs Timer/Counter](http://www.falstad.com/circuit/circuitjs.html?ctz=CQAgzCCMCmC0AcICsA6ALEgnPSSAMeYYkaaATGsnsiGhEnJJAFABKI1aisYZtikEH0gFRBISiQSp1PJOZk8AdiiKh8aiMpl4iSgGEAymWYAZVdR2a1YPNo4gAZgEMANgGdoNPGahI+Vn58tvbULh5eMgrKQeqaYABs6nogRiyKKpDw2hpQCYg6KUY+5lk5mtngdkJQTm6e3swA7lCVgWVCeIg+ALIgSpB8+JQDQzo1ZPLmuojDIDPI43xh9ZEc0SoLc-BKQ4O0IAA67krMAB4gmHiCSAl82PQJUtqUAPL6rHgAghuX1du6ZD7SjHBK+K6UOYQxYFBzhBpRcyYRJUSjIpL+WGCeFrHwZS4oqFqJDAo7uNC-TDE6rQkl8EHuJDg6lo4lLWo4xpIypQnnslYRRr47CQmn5IH0snpGIicS8yGk44lKCiVEquX8uqCxGXcVQvXs7GrIUxTB6mn+CUHY4mYWWqGExXuMDg+0WsawgUI9Z9Xazap+mETeS+hLUOZKMNBgIh-qkNVKeOY4NMvoaf2UdPR6TMNMUNXwfPJmOp-pUhPl4s532RhO1quTJnmRLUYjBFFtmpG7XrC5KQjIBIQJS5W5JF4gAAq+gAcpPlbhhFHICjIFHllrvQvLQlKCukrual7cc1aGaQLAuEJeBerz4WmQb5eCjfDz4LphMOAwF-IJgpGA5AHPm04fHgABCp6RuGSRaDB6wtEoO57oQfBvlB8a7Cq+ZYfe-QorkIhrvMshsCquSwBQAh4OelF3g4aDhg4sjyB0i4gIef5oXu4DMGuE6Hq+2h2L+sg0FIMjid4NCcnh7GQLW8lRnJlrnlh54qcI+yEfseFYSQmaaPGeGEfGOnLKe8njICgzdKeNnWQItYmU5ST6c5lmWgZupQMZp7nt556KHZLRBV0-QBOFmkkTFuQmdQuS5Hep75sFDhpXhlhRfwCEMYgSUhbE7SWmQxHRaVSSKJYZUpeIFWdFlhX1WlzX5nhrV7moZBtZZXX5oMlhxZZqFQPsIhCbl41acIdiaLpw2MTFcGDpNeCpSh+bPqtN43lNF41Yh4pXstyWIbWi0naRiGVLtI03np5b9WtE4+GusEjXR1GLZ9nRoF+8DDeFsBrogSgosD4pydcSTA7WiaULDSR6ZawOVCIm0dFDT5cf0UbA+WelRrcuPUPjFmISiaBYWDSRDYhwLU5TuFQfsEYM+TcaUFw1DwxFuWBtznP8FdEUZqLwv8-GXCg0m1TI-SgK89L-PIf0lrK4T4YBkTSNQediv1nLevjorR2AoTeig2bhVKPrPPnSLtsmzz5aCw9QwBq7dP9DdfAjlzzMtFm2zhVTHNKK7h4RwElB4jETsJkTpIuvi0d1hipK2jEhaioZsyknHmzhRGlaktKKi8xG4x0gcTKp0TAZJqSFKp9XAas6SYKp6z1QJOeNcjMNxGUTh5Yj7HQ8w1RKrD91E8tHtP1EVPp3YQjj4zU+G+rZt9Xo+v91QZUlHEfvF71VjfCUeFCfX4Vi97x9GXDVv4iRtwVUq9xatX5jUEETzcGINJYjD9vGRG-MHb9FrHffmZtcYf3EHpX2Pt14HTLDHDB58eqHUthxfMq87hc3tuOEWQ5f4HgxrrFo5DwBoWxtQji30rxENvIVWhN5-Z0NyruUmLDFo4KYZYEYj1574MhG5HWPD8y5FYcTPCtCo6qwUbtfYiixG8L5rzQO+FabUF3PlEWgxiK7W3ofBe9VTHD3QhY76p8KCk3MRYPQ1gryLTktPfq093G9R2sIbePVcCdQEcE0mp8agSV6tY0JF4bHOKEMdFqGkomVUIpYjmA0CiJNDkYsgWF2jNSGviTJQgAo4xCAcfQrwejfAXEk2CaVeCWDhMaKIxS8kBACmQYmFSDDVO+CwUoHTSkNKwk0rsm4TwWPyV0-JQ0LhMEwIIMAxNbIqBWV+CcoFPg-D6LZbgnFxmwEPCWSyOMtqDGYXZXADTvqcW6TDTivYoBEG4EgPcYAdgXhJMBN4+hC7kTCaubywNKYgFcM4AARtAVwvwfzBGqGgExiKQAAGNXDQGcAAJ1+H9Sqi08VCCwpQdFmKcX4m3ic4mh4SWuAANbzlxbRYiGAYYspAAAEwAJZYugKigALlygA9gAO2YCgEAABVAADgAESFU0EV+ghUAFcRX8ugFi+A4EuX8ocBAEQLyQCQqFfy-lQqAC2tA0TgFSKYAA0keGoABFVs+qQDOsELgWoEBnXCC-MED1EBHy2t9TkA4vqAL0Ftc6pIKI40epUEi-otrWAAFFDA0ADSAc10raAQEQBASVAAFDitQ+AACl7UADFXBculbWoV0q02uAtccSANp23HDAMcNAxwkCHCxQASRFQAN01RqrFLa23uASMcJQg6vgio5QAcWcBq6d87e3HHgIO14WK10btbTu9tZB20dqxUu1d67oCbudKCU9g7TBCoAOZctRSO6VKr+V3ovVew9t7j3uFwI+4DfasX7oA7+ntYH214EXcuqDQHd3AbncBntl7EM3t-ZgdtfbgO7ogwe7DQGFLtpQ7hzD16j0zpXDaeDwGF1YufW+1Frxv1fp-UBjDLH33sf5Zx39A6sVVtrfWxtzbSMdvcGe4D0mQOof7YO0TdaG2uCbb+hjsmRDttwzJhjwmVPifU5JmdenZPmZgxQJTVGkO0YXTJ-DpU93EZozaaT3UbTCf-SRmdj5yM2jPSJmtqmJN3u6Z2mTA6ZMobIHp7zWG3MyY8+ZudRG7M2jQ3k7t8HbO+ZtLF2T2XgtibUxp7jWmcvdpg4BGzPmktkDQ7V50F7IP5Za922TKyEPUcAzOogNoHO8GUyF4z5X+vRa69FxI3aHMJd63e4gs3u2Eba0lz53a9OMR6xl50TWuuUaM2V0zvbKvkgY0i3tsnhO8bYxx794WeOvr4-drjtHKO3f44J7jaXPuvbvZd5jz27sCYe9xp7rGvtg784R+rfXe3SZm86KzsWYN-nbdN3tGGZxCtc-D8kiOmso5tGjvTCmYPxjnUqanSpJCHCU8B+nMmmcDoZx2ztSm2dM7PTTxkHPGfHBxCzpnDH2fM959FsXPO+dc-56zvnAvxfuHg7LpX8vJfc+Fwz6XjZ+e69nSLh91ODfG6l1rhXZvOcW81xLo3yuu1y+F4r6X6uHdK5Vwrqn9vnfm41471X0uPfRa96Lm3Mvrf+4j+7p3Ie9d06V-rnX8ek+Te58ntP0WPe7tt-bk9VumRIs4LRK8txHHCClXKhVSrVXqs1dq3VHF+0cFp9FqQHnjdt8i63qAXem-acF6sBnmhIsGZ78z73nfx+j-bxPsfsnp8j77zaSQQhl9SDQ-opUjfldz477v2fM-R8b-3wvqfTfD9N+PxfxkJ-z+L5v1f+-k-5937Pw-rfRWV+f6kN-1fMmv9r5-6NjN7uAoaj4MZgH9qniaLeSaI6K0JMBoT9TzQ0IEL6L9S6z4isIOIcTEw4GUAmpmqWpwrrg3iMSWBkGco8p8qCqiq+AtjXjtiVQ3jl6chtIxDYHhRPCVShwgAWrOBnBnJjzHT7hsLrDFLnIiFjzEqpDTh8T4DgBRgVLwpVA8Qbh1wxAMFxZxoojaGVJyEACEqQlq0q2K66uODgQ6eAAANMcEOpALYe4EOmQI4UOmAI4RWjYccBWg4d4S4d4b6sYRaqYViuYVihwGIKIIcAAI4doiqiCyjXAxFxEECJE6axGHDxHXCyhMDJGZHXAJEEB5HxFEQFEdoZElE5HpEpFMBES5EVFpFRENGpE5HFGlGlFtHZF1FtFVFFENF1FEQ9EDHwb9G1G1HFEAD8Ux0xUxMR8GYqEq8AkKuqyqwRZh5q4RzSp8KgVhtQzS9hEyfAzhtqAQIAbhEaNQQ6KEwEZxbeNwlxDSZaRxmQw4NQnhNAsIPhZa2aFafsKobxBao0bxe4nUbxbebewaIAFasEsE9gFamQmQP+HqUAX4bq+gKq7g5qFqt2v6kWFODOaGDmFGg66JmJlqOJpGou8m2maO+GCma47aDmWQO27WzJ7gZOv4TOqxIRYRTuYu3J6xuOfJYe6uzAheKoVyKoWECAtQSxKxJhgp4R0mApoRGxW+ypCpqpQpO+1+TIC8s0eQEpnARiH0p8BpEMO8CMx05phC-UlOF0fk+pOEM0dpYiZ83ky03kCi-UtYBiUAHkNCx0vpV43sy8S00M-pjCYZCk70F0AZqCtAWEWYYc-MkcmYBCcSOctAh4WZcwl8+05Yi8f8TpCMbEz0+0kMk8+0ikEZwMloUMQMwCso3AohDZLZq4ppKIUMw88kpp9Zw0VC1EDC98EZdAM0BKTiYZRA70LKk55Z5BRptASCA5lAvA1EaBJpqU4Uy0z8+pLBF0BpOBUMAi+kh5ySoU4o7Q5ZgQeEZog0Q5AQ5s-kJUAUZ5jCmA8YFSy0FS+ZhQKo+5m55Qa8nQI5i005i5nY+ZiQX0rYlZe5wQ4oe04yt5+w4FU5XZ-kNg4wZ8a5uUVIhaagSFagt5HYY0BpkFgMnAnEqoeZ-kR01F4gaAcSd5iZ9wlQKZt59FaIl53sIoaofFHFdF95uonAOiZoYs4lEst56suQklbsz5Cs1A-4nS75n5NIUsT5oUMliAH5XMmlBIGINIpFHM+FdCSl4wGsmFlsOlaglloU-wNIiKvFTl5lnAvFtlslWFhUWw1QCwglQcgIKZSy9IYl7MlwqFGFoUHlNlWsYiyl-AOlboE8QqeQgJxpn4UAKA3SHAKAYgmAkYtw2VWxQ4vEKVuA-0DENEswOVXqUAzAZVSZlVGVMguVog+V3B2VpxCkwQzAQAA). It has two subcircuits (UpDown-Counter and 8bit Comparator) in "normal scale", which limits the zoom factor when the circuit is centered. A custom logic block would be an alternative, but I think it would need a lot of lines with patterns in these cases..

closed 2023-02-03 by pfalstad