sharpie7/circuitjs1 #773: Invalid wires voltage (should be zero)

rayrapetyan, 2022-06-07:
Way to reproduce:
1. Open Circuits -> Basics -> Ohm's law
2. Make sure current is running
3. Click on the wire segments between resistors and the ground. Make sure voltage is zero there.
4. Click on the wire segments between the voltage source and resistors.
Expected: voltage should be zero there too.
Actual: voltage is shown as 5V.

rayrapetyan, 2022-06-07:
Then why it's showing 0 for the bottom wire segments (see step 3)? Real DMM shows zero for both segments btw.

pfalstad, 2022-06-07:
because the bottom segments are at ground.

what are you measuring with the DMM, the voltage between the segments and ground?  or the voltage across the wire segments?

when you mouse over a wire segment, the voltage it's showing is the voltage between the segment and ground.  there's no point in showing you the voltage across the wire segment, because that's always zero.

rayrapetyan, 2022-06-07:
Oh, so you mean in your app one of the probes is always connected to the ground? Is this possible to connect the second probe to any other point in circuit?

pfalstad, 2022-06-07:
when you mouse over a wire or a node, then it shows the voltage compared to ground (which is 0).  You can move the ground node around, but that may cause problems of course.  There's no other way to change the reference voltage.

when you mouse over another component like a resistor, it usually shows the voltage difference across the two terminals (Vd)

To get the voltage difference between two other points, you can create a voltmeter/scope probe by selecting "Draw->Outputs and Labels->Voltmeter/Scope Probe" and then clicking on one point and dragging to another point.  then it will show the difference between those two points.

closed 2022-06-07 by rayrapetyan