sharpie7/circuitjs1 #776: [Feature] Voltage DC Sweep

PedroRodr, 2022-06-13:
First of all, thank you for the incredible circuit simulator, it's very useful!
My feature request is a voltage DC sweep. I want to analyse the behaviour of a circuit from X (V) to Y (V), with a timestep of Z (ms), and I can't find a way of doing that right now.
Sorry if it's a feature that already exists, but I can't find it.

Thanks in advance,
Pedro

pfalstad, 2022-06-15:
You can use a triangle wave source, like in Circuits->Analog/Digital->Flash ADC

If you want it to go in steps, try using a VCVS like so:

https://tinyurl.com/2yyj8jdy

You can edit the VCVS to see the output function.

floor(mod(t/.01, 10))*.2 + 1

.01 is the timestep in seconds (Z).  10 is the number of steps before it starts over.  .2 is the size of each step in volts.  1 is the lowest voltage (X).

PedroRodr, 2022-06-15:
Not exactly what I had in mind, but that'll work! Thank you very much for your help!