sharpie7/circuitjs1 #782: Cannot use analog switch in subcircuit

KevinMLeClairJr, 2022-07-08:
There seems to be limitations with using the "analog switch" in a subcircuit. I had found that all switches "normally open" turn closed and can not be actuated by logic signals at the nodes after being imported into a circuit. Within the subcircuit design everything works fine but outside of it the switches are non-responsive.

This was found during my effort to emulate a CD4051 analog multiplexer/demultiplexed.

pfalstad, 2022-07-08:
It's working for me.  Can you give an example?

KevinMLeClairJr, 2022-07-11:
Sure! This example shows how the analog switch is stuck closed when within a subcircuit - https://tinyurl.com/2q23hzry
Changing the state of the switches latch does nothing and the output reflects the input in both positions - so the normally open switch is stuck closed.

In my analog multiplexer/demultiplexer circuit emulating a CD4051 chip (3 to 8 decoder controlling 8 analog switches) it behaves the same - https://tinyurl.com/2gmqqb9g

pfalstad, 2022-07-11:
It doesn't seem to be stuck closed.  It's just that the switch, when it's open, is still able to transmit voltage (but not current).  The open switch is treated as a large resistor.

https://tinyurl.com/2qar2kre

I added a pulldown resistor to show you that it works.

https://tinyurl.com/2z3tkhmh

Here is the CD4051 with pulldown resistors.

https://tinyurl.com/2ev3bde8

There used to be a similar problem with the tri-state buffer.  A lot of people were complaining so I added an internal pulldown resistor, so that the output is low when the buffer is disabled.  If you can use a tri-state buffer instead, that would fix the issue.