sharpie7/circuitjs1 #788: rotate and flip

tcurdt, 2022-07-28:
Please allow for an easy way to rotate and flip a selection of components. (see kicad for an example)
Recreating a full (sub)circuit just because the orientation should change is rather cumbersome.

pfalstad, 2022-07-28:
if you right click a chip or subcircuit and select Edit, there are checkboxes to let you flip the orientation.

tcurdt, 2022-07-28:
Not really. An especially not for a whole selection of components.
When I right-click I lose the selection and can only edit one.

On a transistor there is "Swap E/C" which really isn't the same thing.
At best it means flipping vertically.

It would be great if there was a "rotate CW", (maybe a "rotate CCW",) a "flip x" and a "flip y".

And applying this on a selection of components would be much appreciated.
I love cicuitjs - but this feels very lacking.

ormaaj, 2022-08-13:
Point reflection of the axis-aligned minimum bounding box of a node selection across the symmetric axis is a very useful and common transformation in schematic capture software, as well as rotation and scaling. It saves lots of time redrawing things.

These would normally be pretty easy but the lack of a uniform way to rotate and reflect symbols is kind of a problem. Kicad symbols have fixed geometry and don't scale or stretch so arbitrary transformations on nodes is easy there. In circuitjs it's pretty much only the 2-node symbols that are drawn in a way that's conducive to this.

esaruoho, 2026-03-03:
Fixed in v3-dev — flip X/Y/XY support for selected groups added.