sharpie7/circuitjs1 #804: Reset issue

Plusred0407, 2022-10-03:
when i press the reset button，crystal frequency waveform is error，It is different from the waveform before reset，The same problem happens when I use crystal electronics

$ 1 5e-11 7.619785657297057 55 5 50 5e-11
r 352 48 352 128 0 11000
r 304 128 304 224 0 68000
r 416 176 416 224 0 1000
w 352 48 416 48 0
w 352 224 416 224 0
O 416 144 464 144 0 0
R 416 48 416 16 0 0 40 5 0 0 0.5
g 416 224 416 240 0 0
t 352 128 416 128 0 1 -3.6976405169273177 -0.7298831605384892 1000 default
r 416 48 416 112 0 100
w 304 128 352 128 0
w 352 224 304 224 0
c 352 128 352 176 0 1e-10 -0.7298831605384892 0.001
c 352 176 352 224 0 1e-10 2.032242642934637 0.001
w 416 176 352 176 0
w 416 176 416 144 0
r 128 128 304 128 0 10
c 128 224 304 224 0 1e-10 0.8901967398765761 0.001
412 128 128 128 224 1 0\s2.87e-11\s0.3331214573955966\s0.001 0\s1e-13\s405.19588754339344\s0.001 0\s0.0025\s0.01509322176680827\s0 0\s6.4
o 5 32 0 12298 4.146903585778256 0.0001 0 1

pfalstad, 2022-10-03:
It's not in error.  It just takes a long time for the crystal oscillations to develop.  By a long time I mean a millisecond or so.  Not very long in real time, but the simulator is not real time so it takes a long while to get going.

Plusred0407, 2022-10-04:
Thanks
I waited for a long time, indeed as you said,

i know how to use it