sharpie7/circuitjs1 #808: Component value scroll skips every other value

mordae, 2022-10-17:
(First of all, thanks for an awesome tool!)

When running on Fedora 36 with `chrome-sandbox`, component scroll skips every other value.
This is with `2022-08-21` release from https://falstad.com/circuit/offline/.

Steps to reproduce:

1. Place a fresh 1k resistor.
2. Scroll down on it to select 1.2k.
3. Instead of 1.2k, 1.5k is selected. 1.2k was clearly skipped.
4. Scroll up to return to 1.2k.
5. 1.2k was skipped again. Resistor is now at 1k.

azambon, 2022-12-21:
I have a similar problem with Windows 11 and Chrome 108.0.5359.125
This appears to be caused by the system's scale setting (in case of Windows, it is found in "System > Display > Scale"). With high resolution displays, Windows suggests a scale setting other than 100%, which also seems to affect the mouse wheel.

azambon, 2022-12-21:
A super-quick search yielded this page: https://stackoverflow.com/questions/1713771/how-to-detect-page-zoom-level-in-all-modern-browsers
I tested the window.devicePixelRatio property both on Chrome and Edge, and in both cases it reports the correct scaling factor of 2.5 for my computer, which matches the 250% factor that is set in the Windows' settings.

So maybe it is enough to divide the MouseWheelEvent's delta value by this factor to fix the issue? But then the browser's zoom level would also affect the result...

azambon, 2023-01-24:
@pfalstad That's MUCH better now. Thank you! 👍 

Although, the issue has not gone away completely, at least for me. For example, if I start with a 3k3 resistor and I scroll slowly to increase the value, some values are still being skipped, namely 5k6, 12k, 27k, 56k, 220k, 470k...
This is still with the default 250% scale factor of my computer.

mordae, 2023-02-20:
For me (Fedora Linux, Firefox, 4K screen with 200% scaling, no browser zoom) it now takes 6 scroll steps to switch to the next value.

pfalstad, 2023-02-20:
It takes 2 for me.  I will probably need to make it configurable to make everyone happy.

esaruoho, 2026-03-03:
Fixed in v3-dev — mouse wheel sensitivity adjusted for component value scrolling.