sharpie7/circuitjs1 #809: LED array - lights fade very slowly

gjwehnes, 2022-10-17:
The LEDs in this component fade _very_ slowly...

Would it be possible to have these fade much quicker, or even instantaneously? This would then allow images to be alternated, which is probably much more in line with how these components would be used in actual circuits.

pfalstad, 2022-10-18:
The LED fade is tied to the update cycle, so if you reduce the simulation speed, they will fade faster (in simulator time).  Obviously this is unrealistic.

esaruoho, 2026-03-03:
Submitted a fix as pfalstad/circuitjs1 PR #275 targeting v3-dev. The brightness decay was 0.99 per frame (1% fade) which took ~7.7 seconds to dim — far too slow for multiplexed displays. Changed to 0.55 for a ~4-frame fade-out.