sharpie7/circuitjs1 #810: how does clock work?

matelot, 2022-10-18:
If I add a clock as an input to one of the circuits and change the clock speed to 1cps how do I get the clock to cycle?

pfalstad, 2022-10-18:
you want 1 cycle per second?  create a clock, right click on it and click Edit, then set the Frequency to 1.

If you're going to make a circuit that slow, you might want to select Options->Other Options and set the timestep to something larger, like 10m.

matelot, 2022-10-18:
Thank you pfalstad, I don't want to sound ungrateful but if you read my question again you will see that I set the clock speed to 1. (1 cps clock per second)
The wire from the clock stays green and doesn't switch off.
[image]

pfalstad, 2022-10-18:
How about the time step, did you update that?

you should probably send me the circuit with File->Export as Link

matelot, 2022-10-18:
Thank you pfalstad changing the time step size did it.

closed 2022-10-18 by pfalstad