sharpie7/circuitjs1 #812: Feature - Running time display and component information

davidllwilliams, 2022-10-19:
I'm intending to create locked circuits for high school students to learn through discovery (but guided) on Moodle. The workflow is a moodle description question containing a iFrame featuring a simulated circuit then the student analyses the circuit and answers a set of questions about that circuit. Howver, everything must be locked so that the student can't change the circuit and settings etc. 

I have used the URL parameters to  hide everything e.g. menu, sidebar, editable .... but please can we have a feature to hide the information box on the bottom right via url parameters? 

In this way I can get them to say calculate the power of a resistor and enter it into a moodle quiz without the simulator spoiling the suprise! Please find attached picture which may help. 
[image]

As I didn't know the name of the infobox (and it is a canvas item), I couldn't search to see if this has already been requested. 
Thanks 

Dave

davidllwilliams, 2022-11-04:
Hi everyone, Quick update, I've looked at code (but struggling to get GWT to install with eclipse on my system) to modify this. 

I've now found the code in CirSim.java line 1668, I basically want a URL parameter to be able to disable/enable this string object so as to stop kids from aquiring the information instead of calculating the answer: 
[image]

Do you think this could be achieved easily by someone out there? If I get the thing to work then I'll attempt it but the install isn't happening at this point in time. 
Thanks
Dave

pfalstad, 2022-11-04:
use hideInfoBox=true

closed 2022-11-04 by pfalstad

davidllwilliams, 2022-11-04:
Absolute legend! Thanks. Dave