sharpie7/circuitjs1 #814: Adding more decoders

LuanHimmlisch, 2022-10-25:
Could you add one of the following decoders? 

- The 74154
- The 74148

Both of these have 8 inputs and 3 outputs

pfalstad, 2023-01-15:
A 74154 is a 4-to-16 line decoder.  You can right-click on a Demultiplexer, click edit, and change the number of select bits to 4.  The one thing we are missing is the G1 and G2 strobe inputs.  Do you need those?

The 74148 is a 8-to-3 priority encoder.  Here is one I made with custom logic: https://tinyurl.com/2f3wr5w4

LuanHimmlisch, 2023-01-18:
@pfalstad thanks a lot, I mainly opened this issue for anyone else having the problem, like a teacher, since this was required in a classroom and I always like to share open-source alternatives

closed 2023-01-18 by LuanHimmlisch