sharpie7/circuitjs1 #823: Monotonically increasing current without limit

zpillio, 2022-12-04:
As the title says there are cases when the current increases monotonically without limits.

Here is the circuit:

[wtf.circuitjs.txt](https://github.com/sharpie7/circuitjs1/files/10149071/wtf.circuitjs.txt)

I haven't tried to reproduce it in a simpler circuit, but it's related to the fact that I accidentally swapped drain and source of mosfets.

pfalstad, 2022-12-04:
Hi,

It looks like the current is flowing from the DC source, through the MOSFET body diode, then through an inductor, then through another MOSFET body diode, and back to the source.  There's no resistors in this path, and the inductor and body diodes are ideal (with no resistance), so it makes sense that the current would slowly increase without bound.

I think it's equivalent to this: https://tinyurl.com/2pp4pl2x

zpillio, 2022-12-04:
Thank you for the explanation. So those components have no resistance in this model, and this is an expected behavior if I understand correctly.

closed 2022-12-04 by zpillio