sharpie7/circuitjs1 #824: Getting "Matrix Error" when using a switch with a subcircuit

kylepaulsen, 2022-12-12:
I might be missing something obvious, not sure. I'm tying to simulate a NAND gate as if it were on a breadboard.

[image]

Main circuit:
https://tinyurl.com/2la74rgh

Subcircuit:
https://tinyurl.com/2p3dwqgj

If I open either switch on the main circuit, the simulation stops and I get a "Matrix Error" message on the bottom. Any idea why? Thanks.

pfalstad, 2022-12-12:
Hi, if either switch is open, then the voltage of A or B is undefined.  A and B are only connected to MOSFET gates, which don't help define the voltage of those lines.  If you don't use a subcircuit, then the simulator is smart enough to notice that A and B are undefined and so it will set them to 0.  But if you do, then it gets confused.  You should probably use logic inputs for A and B instead so that the voltage will always be set.

kylepaulsen, 2022-12-12:
Thanks for the reply. I saw the logic inputs, but I was trying to mimic a circuit in real life for learning. If the subcircuit used NPN transistors instead would that have worked?

pfalstad, 2022-12-12:
yes that would have worked.  or, you could add a 1Meg resistor from each input to ground.

closed 2022-12-12 by kylepaulsen