sharpie7/circuitjs1 #831: the fucking ram doesn't turn off

18001767679, 2022-12-30:
it doesn't turn off output when OE' is on!

18001767679, 2022-12-30:
it needs to be pulled down >:(

pfalstad, 2022-12-30:
So pull it down.

18001767679, 2022-12-30:
oh ok :(
a bit uncomfortable seeing that much current in circuit lol

18001767679, 2022-12-30:
also the latch latches on RISING EDGE????WHAAAA?
I need some ram and latches for my 6 bit subleq cpu

18001767679, 2022-12-30:
i need to
a)make something that latches on high instead of rising edge
b)make rising edges

pfalstad, 2022-12-30:
I thought that the point of the output enable on the RAM chip is to so you can connect something else to the bus while the output is disabled.  But I suppose you might need to have the RAM output actual zeros when output is disabled.  Is that what you need?  I could fix that.

So you need a transparent latch?

18001767679, 2022-12-30:
cannot access tinyurl :(
can u paste text here?
Yes I need it to make zeros or it will mess up the bus

pfalstad, 2022-12-30:
You can't access tinyurl? Why not?

http://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgoqoQFMBaMMAKAEIQAXAJwEMA7AM4AHXtzr9OAHUEAbXpwDGACyggAkpAA06sDpQ7sWgMKyA1iADy2y3ssHL2LnyGjxkmfKWqAggCFjABEwKQBHfyCpfgB+WOjIGQjAsKSoljQ8EBQEQnBCKlwAFjyqKh4BETEJaTkFFWpSihYAGSyckqyUYrB8tQgAM15ZQToGlrbcsGIUTu7ptSpB4dGkSHHs3JQu2Z2qAaGRsdaNiZ2TvZAlw9X19pQ8TK3i+8zGq5WoFgBZEGwEGZ6BTw3V6VBQ0AQ31+-3A82wwNhMzBEKhfxmT1+CIxyMhPzRp3hz3aOJYQA

Usually if the RAM is disabled, then something else is driving the bus, so you don't need or want pulldowns..

18001767679, 2022-12-30:
If something else is NOT using the bus,i might want to put the ram onto the bus(not when something else IS using the bus)

18001767679, 2022-12-30:
Thank you,although i already pulled the bus down

closed 2022-12-31 by 18001767679