sharpie7/circuitjs1 #832: Why so laggy

18001767679, 2022-12-30:
I want to make something like an cpu.
But if i put enough ram modules(i need 3 16*16 ones(i.e 16 address and 16 data pins))it lags a lot
If I then put a lot of registers and wires on a long and thick bus,it will lag pretty much.
I have a schematic of a 6 bit cpu that has 3 6*6 ram modules and a lot of registers and lags pretty much.

pfalstad, 2022-12-30:
Yeah you might run into problems with a circuit that big.  You might want to use a logic-based simulator.

18001767679, 2022-12-30:
a logic based simulator?(i just used your ram modules that lack pulldowns on a seperate thread,but they still lagged,although less than my own ram modules)
ps:logic.ly gets laggy at even smaller scales

pfalstad, 2022-12-30:
Really, it's even more laggy?  Well, you could play around with the timestep.  You can set the timestep using Options->Other Options.  Make it as large as possible.  Take the clock period (1/clock frequency), divide by 2, and divide by the number of steps needed to simulate your circuit.  Like if you have gates 5 levels deep, then divide by 5.  Or you could just experiment and see how large you can make it without introducing errors into your circuit.

18001767679, 2022-12-30:
It's not laggy as if it would run at low tick rate,its laggy as if you cannot drag it freely(fps plummets on dragging stuff)

18001767679, 2022-12-30:
Hello?

pfalstad, 2022-12-30:
So it's laggy when dragging?  Even if the simulation is stopped?

I don't think I have any circuits that big, could you post it?

18001767679, 2022-12-31:
https://replit.com/@imcute-aaaa/circuitjs-schematics#6%20bit%20finished%20(1).txt

18001767679, 2022-12-31:
It lags when you don't stop the simulation
And it lags when you drag parts

18001767679, 2022-12-31:
Logisim worked fine.(partially cuz they have wider wires/buses natively)

pfalstad, 2023-01-08:
Sorry, it's not designed for huge digital circuits.

closed 2023-01-08 by pfalstad