sharpie7/circuitjs1 #833: [macOS] version number stuck at 9.3.2

core-code, 2023-01-10:
every release posted here ( [qiyimedia](https://www.falstad.com/circuit/offline/ )  in the past years always had the very same version number: 9.3.2

is this on purpose? shouldn't the version number (CFBundleVersion + CFBundleShortVersionString) increase with every new release?

pfalstad, 2023-01-10:
Oh the version file in the dmg?  I'll get rid of that.  That's old.

To get the version, use File->About

core-code, 2023-01-10:
no i am referring to the real version number:
[image]
[image]
[image]

pfalstad, 2023-01-10:
ok, Il'l fix that.

core-code, 2023-02-05:
it has now changed to "2.7.6js"