sharpie7/circuitjs1 #836: Feature request : more op amp models?

MichaelWeaser, 2023-01-18:
I know the 741 and the Lm324 is really common used op amps and I know the 741 is used as an introduction to op amps , which is why probably those op amps are the only ones listed. 

Could you eventually add other op amps? I am thinking like maybe like a TL071 FET op amp? I am thinking maybe other op amps like NE5532 audio op amp ,  OP07 classic precision op amp. 

The only reason why I think other op amp models should be available is that the 741 really isn't used anymore in designs and the LM324 usually isn't used either.

thephantom1492, 2023-04-04:
I have also an idea, about a parametric opamp.
It shouln't be too hard, and should allow the user to simulate virtually any opamp with only a few parameters.
I think that the parameters could be:
- input to each rail voltage, ex: VCC -2V, VDD + 2V
- Output to each rail voltage, ex: VCC -1V, VDD + 1V
- Gain bandwidth, ex 1M
- Output max current, ex +25mA, -25mA (allow 0 for open-collector)
- internal consumption, ex 1.7mA
(those values are from https://www.ti.com/product/LM741)

I think that with only those parameters, virtually all opamp can be simulated with a good enough precision for most simulations.

MichaelWeaser, 2023-04-05:
@thephantom1492  What about the option to upload a op amp model? Maybe the option of uploading a op amp spice model, could allow different op amps?

thephantom1492, 2023-04-05:
AFAIK, falstad do not use spice model, but hard coded modules. But that I am not sure, maybe when the dev team answer they could highlight it, but until then, I do not know.