sharpie7/circuitjs1 #837: Copy of custom logic do not work

hjanetzek, 2023-01-20:
In this circuit only the first (rightmost) custom logic instance is evaluated correctly. The copies don't work, not even when creating a new instance and copying the model configuration. 

[custom-logic-bug.zip](https://github.com/sharpie7/circuitjs1/files/10465253/custom-logic-bug.zip)

Awesome work btw! this program really helps me getting started with electronics (:

pfalstad, 2023-01-20:
Hi, you have two many question marks in the first line of the model.  You have "??????????1" which is 10 question marks.  So the 1 matches the first output (0).  So as soon as output 0 becomes high, then the outputs go into a high-impedance state and stays there no matter what the inputs are.

Also the rightmost custom logic instance has a slightly different model where the outputs are all 0, not _.  So it doesn't have that problem.

hjanetzek, 2023-01-20:
Oh thank you! :)

closed 2023-02-01 by pfalstad