sharpie7/circuitjs1 #842: Reading error with Diodes in series with resistor

alexabrev, 2023-02-01:
I simulated two diodes in series with the cathodes pointed to the positive side of the loop and a resistor connected in sequence. The resulting voltage on the anode side of the resistor should be zero, but it has voltage. Could someone explain to me if I used this simulation in the wrong way or if there is some bug in this type of circuit assembly? Thanks

$ 1 0.000005 10.20027730826997 50 5 50 5e-11
w 1216 0 1392 0 0
w 1392 80 1392 160 2
w 1216 144 1216 528 0
w 1216 528 1392 528 0
r 1392 288 1392 368 0 2050
d 1392 240 1392 160 2 default
d 1392 80 1392 0 2 default
w 1392 240 1392 288 2
w 1392 368 1392 528 2
v 1216 144 1216 0 0 0 40 12 0 0 0.5
g 1216 528 1136 528 0 0

alexabrev, 2023-02-01:
Between D1 and D2, if I'm not totally wrong, the value should be 0 V, not 6 V as it appears, right?

Thank you very much! Your project is amazing!

pfalstad, 2023-02-01:
the current through each diode is the same, so the voltage across each one should be the same.  It doesn't make sense for one diode to have 12 V across it and the other one to have 0 V across it.  Each diode has about 6 V across it.  That's why the voltage is 6V in the middle.

alexabrev, 2023-02-02:
Perfect! Thank you!

$ 1 0.000005 10.20027730826997 50 5 50 5e-11
w 1168 0 1344 0 0
w 1344 0 1344 16 2
w 1168 144 1168 528 0
w 1168 528 1344 528 0
r 1344 256 1344 336 0 1000
d 1344 208 1344 128 2 default
d 1344 128 1344 16 2 default
w 1344 208 1344 256 2
w 1344 336 1344 528 2
v 1168 144 1168 0 0 0 40 12 0 0 0.5
g 1168 528 1088 528 0 0
o 6 64 0 4099 10 0.00009765625 0 2 6 3
o 5 64 0 4099 10 0.00009765625 1 2 5 3

closed 2023-02-25 by pfalstad