sharpie7/circuitjs1 #843: Question/Bug-Correct Usage of setTimeStep()

JustusE20, 2023-02-07:
I am running the simulator in an iframe and want my webpage to control the size of time steps in the simulation. That's why I am executing simulation.setTimeStep(timeStepSize) on simulation.ontimestep, which does change the time step size. However, when the time step size set by the webpage deviates from the value in the simulation settings, the messages "timestep up = 0.000005 at 0.073..." and "converged after 0 iterations, timeStep = 0.000004" are being issued constantly in the console. It seems as if my time step value is being overwritten by the settings value. Is there a way to avoid this overwrite or at least the console output?

pfalstad, 2023-02-08:
It looks like a bug in simulation.setTimeStep().  As a workaround, try setting the timestep in Options->Other Options... to something very small.  Then use setTimeStep() to something larger.  As long as it's larger, you shouldn't see those messages.  (I haven't tested that, but it's worth a try.)

JustusE20, 2023-02-08:
The workaround works in preventing the console outputs, but then the scopes in the simulation get incredibly fast.

JustusE20, 2023-02-14:
I also noticed that voltage values get falsified, when the values of simulation.setTimeStep and the time step size setting differ. To test this, I used the A/C Circuits Capacitor circuit.

closed 2024-11-29 by pfalstad