sharpie7/circuitjs1 #844: Separated transistor being affected by switching.

PernelSned, 2023-02-10:
I have created an oscillator, however I am facing an issue, the circuit is affected by other transistors which are not connected. In this example there is a transistor which has collector connected 100 ohm to 5V and base + emitter connected to gnd. This should prevent the transistor from switching however I am seeing fast voltage drops on the collector.

`$ 1 0.000005 0.14549914146182014 50 5 50 5e-11
r 1360 1216 1360 1280 0 100
w 1360 1328 1360 1312 0
g 1360 1328 1360 1344 0 0
R 1360 1216 1360 1184 0 0 40 5 0 0 0.5
R 2096 864 2096 832 0 0 40 5 0 0 0.5
g 2096 1152 2096 1168 0 0
g 2048 1152 2048 1168 0 0
g 1920 1152 1920 1168 0 0
g 1872 1152 1872 1168 0 0
g 1792 1152 1792 1168 0 0
g 1744 1152 1744 1168 0 0
g 1664 1152 1664 1168 0 0
g 1568 1152 1568 1168 0 0
g 1424 1152 1424 1168 0 0
g 1312 1120 1312 1136 0 0
g 1216 1120 1216 1136 0 0
g 1168 1120 1168 1136 0 0
R 2048 864 2048 832 0 0 40 5 0 0 0.5
R 1920 864 1920 832 0 0 40 5 0 0 0.5
R 1872 864 1872 832 0 0 40 5 0 0 0.5
R 1792 864 1792 832 0 0 40 5 0 0 0.5
R 1744 864 1744 832 0 0 40 5 0 0 0.5
R 1696 864 1696 832 0 0 40 5 0 0 0.5
R 1664 864 1664 832 0 0 40 5 0 0 0.5
R 1568 864 1568 832 0 0 40 5 0 0 0.5
R 1504 864 1504 832 0 0 40 5 0 0 0.5
R 1456 864 1456 832 0 0 40 5 0 0 0.5
R 1216 832 1216 800 0 0 40 5 0 0 0.5
R 1168 832 1168 800 0 0 40 5 0 0 0.5
w 1328 1328 1360 1328 0
w 1328 1296 1328 1328 0
r 1072 928 1072 1008 0 390
r 1664 944 1664 864 0 1000
t 1328 1296 1360 1296 0 1 -4.99999999898 0 100 default
w 1072 736 1072 928 0
w 2128 736 1072 736 0
w 2128 1072 2128 736 0
r 1216 1008 1312 1008 0 1000
c 1312 1008 1312 1120 0 3.3e-8 1.573132192627469 0.001
R 1104 832 1104 800 0 0 40 5 0 0 0.5
r 1952 1072 2016 1072 0 6800
w 1952 1184 1616 1184 0
w 1952 1072 1952 1184 0
w 2096 1104 2096 1152 0
t 2016 1072 2048 1072 0 1 0.5668890494985303 0.6281455312639911 100 default
r 2048 864 2048 1024 0 1500
r 2048 1088 2048 1152 0 1000
t 2048 1088 2096 1088 0 1 0.6195430695209672 0.6421575239125784 100 default
d 2096 1040 2096 1072 2 default
t 2048 1024 2096 1024 0 1 -4.294951124588576 0.4739625887144016 100 default
w 2048 1024 2048 1056 0
r 2096 1008 2096 864 0 180
w 1088 1008 1072 1008 0
r 1312 1008 1376 1008 0 22000
t 1136 1008 1168 1008 0 1 -4.53537230643511 0.4267958724738971 100 default
r 1168 832 1168 960 0 1500
r 1168 1024 1168 1120 0 1000
t 1168 1024 1216 1024 0 1 -3.8542492862517617 0.001481060773326552 100 default
w 1216 1040 1216 1120 0
d 1216 976 1216 1008 2 default
t 1168 960 1216 960 0 1 0.370418769223515 0.6166549032721429 100 default
w 1168 960 1168 992 0
r 1216 944 1216 832 0 180
t 1104 976 1104 1008 0 1 0.5949011320786273 0.6128291894100835 100 default
r 1104 976 1104 832 0 4000
w 1136 1008 1120 1008 0
R 1392 864 1392 832 0 0 40 5 0 0 0.5
w 1504 1104 1504 1088 0
w 1472 1088 1472 1120 0
w 1472 1088 1504 1088 0
w 1424 1104 1424 1024 0
w 1504 1152 1504 1136 0
w 1424 1136 1424 1152 0
r 1504 1088 1504 1024 0 10000
t 1472 1120 1504 1120 0 1 0 0.5739693453606328 100 default
t 1472 1120 1424 1120 0 1 -0.5883602417302444 0.5739693453606328 100 default
t 1376 1008 1392 1008 0 1 -3.4191913531317457 0.41063000530894 100 default
t 1504 1008 1456 1008 0 1 0.5196691861895626 0.5766595460685193 100 default
w 1424 1024 1392 1024 0
w 1424 1024 1456 1024 0
r 1392 992 1392 864 0 10000
r 1456 960 1456 864 0 10000
w 1504 1024 1504 864 0
r 1568 864 1568 944 0 1000
r 1568 1008 1568 1152 0 1000
r 1504 1008 1568 1008 0 22000
r 1616 1008 1616 1184 0 680
w 1616 1008 1568 1008 0
r 1664 976 1664 1152 0 2200
t 1632 960 1664 960 0 1 -3.494958059738531 0.5637369991286241 100 default
r 1568 960 1632 960 0 2200
w 1456 960 1456 992 0
w 1568 944 1568 1008 0
w 1456 960 1568 960 0
w 2096 1072 2128 1072 0
d 1792 960 1792 1072 2 1N4148
r 1920 1008 1920 864 0 180
w 1872 1024 1872 1056 0
t 1872 1024 1920 1024 0 1 0.08576396999157865 0.6049626140290263 100 default
d 1920 1040 1920 1072 2 default
t 1872 1088 1920 1088 0 1 -3.6520063301510217 0.10203526938812933 100 default
r 1872 1088 1872 1152 0 1000
r 1872 864 1872 1024 0 1500
t 1792 1072 1872 1072 0 1 -4.18858329223225 0.5362710403935388 100 default
r 1696 960 1696 864 0 1800
r 1744 944 1744 864 0 1100
r 1744 992 1744 1152 0 390
w 1744 976 1744 992 0
t 1664 976 1696 976 0 1 -1.7120654367416748 -1.0712051341513766 100 default
t 1696 960 1744 960 0 1 0.6044206448997689 0.6408603025902981 100 default
r 1792 1072 1792 1152 0 2000
t 1744 944 1792 944 0 1 -2.9262060670603094 0.5658480049807855 100 default
r 1792 864 1792 928 0 1000
w 1744 992 1696 992 0
w 1920 1104 1920 1152 0
g 1504 1152 1504 1168 0 0
w 1952 1072 1920 1072 0
o 33 1 6 4354 5 0.1 0 1
`

pfalstad, 2023-02-10:
Go to Options->Other Options and turn "auto-adjust timestep" on.  Then the problem goes away.

I should probably make that the default.

When the simulator is having difficulty converging, we use some tricks that affect all transistors in the circuit to get it to converge.  That causes that transistor to turn on.  I recently added auto-adjusting the timestep, which is a better way to get circuits to converge.

PernelSned, 2023-02-10:
That's great, thanks for making this software, I love it!

closed 2023-02-10 by PernelSned