sharpie7/circuitjs1 #845: Adding a capacitor says "capacitor loop with no resistance!"

Killeroid17, 2023-02-10:
This has never happened before...
Have been always adding capacitors in parallel with power supplies or in circuit anything else, and never had this error.
tried to modify all component's values but always the same...
this as an example of simplest simulation cannot be done not to mention more advanced ones...
**Please** put an 'Archived older versions' in the 'Standalone (offline) versions.' section...

[image]

pfalstad, 2023-02-10:
This has been there forever.  See #634.

Killeroid17, 2023-02-10:
> This has been there forever. See #634.

yes already wasted my time trying SEVA77's older version's and yes, but I know long time ago I applied in parallel a cap and didn't had problems I mean I've used caps in any combination without this error,
can't this be implemented so it's general wires to have it's own resistance ? simulating this in real life model won't give you any *SPARK*... so what can you do about it to avoid getting error on this trivial connection ?

pfalstad, 2023-02-10:
Changing wires to have their own resistance would slow down the simulator quite a lot.

I could probably fix this, I will look into it some more.

Killeroid17, 2023-02-11:
> Changing wires to have their own resistance would slow down the simulator quite a lot.
> 
> I could probably fix this, I will look into it some more.

or at least remove the error and leave the project running without being halted / stopped

closed 2026-05-03 by pfalstad