sharpie7/circuitjs1 #847: Simulating resistor failed

Falcon120864, 2023-02-17:
Hi,

In the circuit given after this message I tried to simulate resistor R1 (140 ohm)  with active current/voltage générator R2 (using the fact that R=U/I) but one can see using I1 and I2 (I1 on and  I2 off  : R2, I1 off and I2 on : R1) that AOP's output is not the same. Is it a bug or did I make a mistake ? 
 
  
$ 65 0.0000049999999999999996 10.634267539816555 50 5 50 5e-11
a 224 256 352 256 8 5 0 1000000 0.13961011721386157 0.1396102103025754 1000000
R 144 64 144 48 0 0 40 5 0 0 0.5
r 192 96 192 176 0 49100
r 224 192 352 192 0 60400
r 224 320 352 320 0 105800
r 192 320 192 384 0 1000
w 192 240 192 176 0
w 192 240 192 320 0
w 224 240 224 192 0
w 224 320 224 272 0
r 80 96 80 176 0 4990
w 80 176 80 256 2
w 144 448 192 448 0
w 80 96 144 96 0
w 144 64 144 96 0
w 144 96 192 96 0
g 192 448 192 464 0 0
w 352 256 352 320 0
w 352 256 352 192 0
w 352 256 432 256 2
213 -16 320 64 320 0 2 (a-b)/140
w 80 272 -16 272 0
w -16 272 -16 320 2
w 80 448 -16 448 0
w -16 432 -16 352 0
r 144 304 144 384 0 140
w 80 320 80 272 1
s 144 448 144 384 0 0 false
w 192 384 192 448 0
w 80 448 144 448 0
w 80 432 80 448 0
w -16 432 -16 448 0
s 80 432 80 368 0 1 false
w 80 368 80 352 0
w 80 272 80 256 0
w 192 240 224 240 0
w 144 304 144 272 0
w 80 272 144 272 0
w 144 272 176 272 0
w 176 272 176 256 0
w 176 256 208 256 0
w 208 256 208 272 0
w 208 272 224 272 0
x 32 407 52 410 4 24 I1
x 109 420 129 423 4 24 I2
x 102 349 132 352 4 24 R1
x 214 365 244 368 4 24 R1
x -57 341 -27 344 4 24 R2

pfalstad, 2023-02-17:
You need a minus sign in the output function.

https://tinyurl.com/2q57tkzp

Falcon120864, 2023-02-17:
Ok, I just found it before reading your answer, your sign convention is not the same as mine. Ok 

Thanck you.

Falcon120864, 2023-02-17:
> 

By the way I would like to have a resistor voltage driven active generator as new component.

pfalstad, 2023-02-17:
what do you mean, "resistor voltage driven active generator" ?

Falcon120864, 2023-02-17:
> 

A resistor between node C and D like Rcd = f(A,B)  with A and B two potentials (a quadripole like your other active generators with A-B has input and C-D as output).

closed 2023-04-09 by pfalstad