sharpie7/circuitjs1 #849: Feature request: divisible relay symbol

CodeIgnoramus, 2023-02-28:
For purposes of replicating and demonstrating the behavior of ladder logic/ relay logic diagrams, it would be extremely useful to be able to place a relay coil symbol in one place and link it to multiple NO and NC contacts elsewhere in the diagram. And if you could show their relationship, for example by highlighting all of the linked symbols when one is clicked on, that would be icing on the cake. Also, I am a total GitHub newbie, so forgive me if this is verboten, but I am willing to pay to have this feature added.

pfalstad, 2024-10-24:
use "Add Relay Coil" and "Add Relay Contact"

closed 2024-10-24 by pfalstad

CodeIgnoramus, 2024-10-24:
The relay contacts seem to pass current regardless of their position.

https://ci3.googleusercontent.com/meips/ADKq_NaYKP47Zr-93ikgRn5aSXSwar-T-fZrLTWM5n3q6ObidxXZxJ6MtzqX3R3erpYlpktj7bdiUNGOMSy7ZfXrSLSqhRjbl_izo7g1LWTr2oFT9vFvztfGVt0TScFRKEiYNkbrQq_hpNw8cdYxfGrraon-hs5m8ENMcnQE5dagl7HSo9PO7868i2DnD7baalm776xmkf2NupCTHJ9MaeVI6OMNyHL47_jN6E55SlveVTCx0294GI4z2lc=s0-d-e1-ft#https://github.com/notifications/beacon/A6GC6RP6RA4F2MQY5KBTW4LZ5BNXZA5CNFSM6AAAAABQQCPSJ6WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTURCXOBQ.gif>Message
ID: ***@***.***>

pfalstad, 2024-10-24:
Here is an example

https://tinyurl.com/276x2vv7