sharpie7/circuitjs1 #852: Subcircuit Error Message Text Locking Glitch

Danielboy12345, 2023-03-22:
This only occurs on the Standalone-Offline addition.

When using the "Create Subcircuit" feature under File and there is an error present with the subcricuit, such as a disconnected node or no labeled nodes, the error message pops up as usual. After you close it, however, none of the shortcut keys work anymore nor can you change any parameters of placed components. Whenever you go to edit the components, clicking on the textbox does not make the "|" symbol appear nor does the box get highlighted. You can select the parameters with a cursor drag and delete them, but you cannot type at all. I think it may have something to do with the error message popup, but I am unsure. 

While typing this, I discovered a fix. By clicking out of the standalone window and back into it, it restores the ability to use shortcuts and to type in parameters. The error message popup is most likely not properly refocusing the window, causing problems with the popups of the components.

pfalstad, 2023-04-09:
Is this the offline version?  This sounds like #596.

Danielboy12345, 2023-04-09:
It is the offline version.