sharpie7/circuitjs1 #855: In the startIteration() method in the LampElm class, how are these formulas and parameters obtained?

306Vulcan, 2023-04-05:
In the startIteration() method in the LampElm class, how are these formulas and parameters obtained?
`
     void startIteration() {
	    // based on http://www.intusoft.com/nlpdf/nl11.pdf
	    double nom_r = nom_v*nom_v/nom_pow;
	    // this formula doesn't work for values over 5390
	    double tp = (temp > 5390) ? 5390 : temp;
	    resistance = nom_r*(1.26104 -
				4.90662*Math.sqrt(17.1839/tp - 0.00318794) -
				7.8569/(tp - 187.56));
	    double cap = 1.57e-4*nom_pow;
	    double capw = cap * warmTime/.4;
	    double capc = cap * coolTime/.4;
	    //System.out.println(nom_r + " " + (resistance/nom_r));
	    temp += getPower()*sim.timeStep/capw;
	    double cr = 2600/nom_pow;
	    temp -= sim.timeStep*(temp-roomTemp)/(capc*cr);
//	    sim.console("lampsi " + temp + " " + capc + " " + nom_pow);
	}
`

306Vulcan, 2023-04-05:
can you help me

billcollis, 2023-04-05:
looks like it could be similar to a thermistor

306Vulcan, 2023-04-05:
> looks like it could be similar to a thermistor

Hello, do you have any reference materials?

billcollis, 2023-04-05:
maybe try this?
https://www.eevblog.com/forum/projects/electrical-model-for-incandescent-lamp/
or search for PTC thermistor graph/model