sharpie7/circuitjs1 #860: question about sub circuits

iccor56, 2023-04-17:
is there a way to remove sub circuits from the list of subcircuits in the draw menu?

in the absence of name changing or updating ability, remove and replace is my only method of changing sub circuits but the menu never forgets the old ones.

Vadimatorik, 2023-07-29:
When I want to start a new project. I am deleting app settings. This clears the list of subcircuits. For Linux (Ubuntu 20.04):
```
rm -R "/home/$USER/.config/Circuit JS1"
```
However, this is a bad way. When you open an old project again, all old subcircuits are added to the list again.
I also want to know: is it planned to add this functionality?

pfalstad, 2023-08-05:
Unfortunately right now, the only way to do this is:

Options->Toggle Dev Tools
Click the >>, select Application
Expand Local Storage, select file://
find the offending subcircuit: key, right click on it, Delete

That's for the offline version.  For the online version, if you're using Chrome:

View->Developer->Developer Tools
Click the >>, select Application
Expand Local Storage, select http://www.falstad.com/
find the offending subcircuit: key, right click on it, Delete

This is a dup of #700.

closed 2023-08-05 by pfalstad