sharpie7/circuitjs1 #867: How to simulate a depletion mode MOSFET like the RFP15P05

alien999999999, 2023-05-19:
I tried calculating the beta and i came up with 1.111 for this one.
So i tried putting a P-channel mosfet, set 1.111 as beta, threshold -4V ; but connecting gate to 3.3V does not seem to pinch the voltage; but in reality it does work; unless i'm doing something wrong, of course: https://tinyurl.com/2q27ehfy

vindicatorr, 2023-07-07:
You're farther along than I am (I'm only just now becoming aware of depletion/enhancement), but I found the older closed issue https://github.com/sharpie7/circuitjs1/issues/301 and you can pull the depletion FET code from there.

I'm just wondering where it even came from.
When I export text (after removing everything but the fet), I can see the "f" is common ("(_**f**_)et"?), and the difference (other than position and fet values) is the 5th number (32 (enhancement) vs 9 (depletion)).

Again, I don't know where it comes from yet, and why we can't add it from the menu, or how much else we're missing out on that is already built-in but hidden.

Wait, that "9" wasn't enough. I had to change the 6th number to "-1" from "1" which I see was the `threshold voltage`. So negative whatever that number is.

Trying it again bare:
N/Enhancement
```
$ 1 0.000005 10.20027730826997 50 5 50 5e-11
f 288 144 288 192 32 1.5 3
```
To:
N/Depletion
```
$ 1 0.000005 10.20027730826997 50 5 50 5e-11
f 288 144 288 192 9 -1.5 3
```

Nope, that wasn't right. The arrow direction is different. Trying "8" instead of "9" (must've been "P" fet). Okay, that looks right now (https://circuitjournal.com/which-mosfet-should-you-use-with-arduino).

Mind you, I have no clue whether or not the underlying magic of falstad does what depletion is meant to do.

pfalstad, 2023-07-07:
the 5th number is a set of flags.

32 indicates a body diode.  Every new MOSFET you create has one by default.  You can turn that off though.
1 is "P" type
8 is flipped (Swap D/S)
9 = 1+8 = flipped and "P" type

to make an N depletion mode mosfet, just use a negative threshold.
to make a P depletion mode mosfet, use a positive threshold.

pfalstad, 2023-07-07:
Sorry, [alien999999999](https://github.com/alien999999999), for not responding sooner.  The MOSFET model in this simulator is a very simple model.  You're not going to get very good agreement with specific devices.  In the future I might bring over a more complex model from spice.