sharpie7/circuitjs1 #871: Explanation of simulation parameters

nithinmurali, 2023-05-27:
what is the meaning of simulation speed and time step? and what is the impact of these parameters in the simulation?

My understanding is that timestep is the amount in which the circuit time is incremented in each iteration (some kind of internal loop). and sim speed controls how many times iterations are run in a second ? But this doesn't seem to be correct.

Can you please explain with an example? 
suppose, if I want to make the simulation 1/4th realtime. what values should I put for sim-speed and time-step and why?
suppose, if I want to make the simulation 1/2th realtime. what values should I put for sim-speed and time-step and why?

pfalstad, 2023-05-27:
yes, the timestep is the amount which the circuit time is incremented in each iteration.
the sim speed controls roughly how many iterations are run per second.

The timestep is not determined by how fast you want the simulation to run, though.  It's more determined by the behavior of the circuit and how much accuracy you want.  For example, if you create high-frequency voltage sources, the simulator will set the timestep to 1/(32*frequency) to make sure that the high frequency oscillations can be accurately represented and simulated.  If you have a 10khz signal in the circuit and the timestep is set to something larger than 100usec, then it won't work.  You can't even represent a 10khz oscillating signal with timesteps that large.

You can't really set the sim speed to a particular value.  It's a slider.

If the simulation speed is close to realtime then the simulator will display the relative speed in the lower right hand corner (for example, "(.5x)" means 1/2 realtime)

nithinmurali, 2023-06-08:
Thank you that makes sense. Is there a reason the app doesn't provide a way to set the exact sim speed (input box) so that user can set a speed relative to real time. Like 0.001x realtime.

pfalstad, 2023-06-08:
No, that might be worth adding.  I never needed that (until recently) so I just used a slider because it's easier to adjust.