sharpie7/circuitjs1 #874: SVG export incorrectly shifts position of undocked scopes, and also fails to export docked scope

ericfont, 2023-06-10:
[Here](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BOJyWoeAHNATAdgwmAKwkYBsYALLkdiEekSJQMz0CmAtGGAFABu4MHWEZwCEdjGQQbbDJhsFUaEV4BzZnBDyxlbdmyUoUXgCUtMzizosyY63WX7Z0JSZhqA7pZAYZ+jI8Irw+geCUxuG64LwALrL24FK+oibc4G5k2GC4cAxElGR22GQgnJ4IkAhkebgsSCyUDGVg2gAm7ABmAIYArgA2caGydslidmJpfABOOhhTKVJiMUHwcBrzK5ArCzpGHrwA8r7+iWJnbHxh2o6pZZC8AMbikitErSkKsOtof8jlMCwXCUBYYIokMisexMR6aYLjEA1REKXhzcJgMrhM5BIgbHzYD6I5Yo45bYl7BFXEYsMZpElpR4+WlYzHkyhsx4nEkcso87TUglE3nszlo3wsSBRAVJXEbeHCZhs5Ei1EvMASJV84WclTrar-P6A4GRepgfy4EFS6QjaLCgVS0wncKTc5u6maHnIklERWouY8tleh7gfU0sYi10ivgJKNs8KqwEqS2mljmyApqW4dAVaBkSB4YpVZogli4WqhmSdXqDYY3AI2XyUZHXCVJBPaPgWO1lV2EkMBGQsNweFRqTQJlXaX1OUyCBHRzWJkTaRSjzy2mV6KFux7oneuj5D2WV-H0fNu7LGV1MkBXy8HQzGW-3p8gXDyfbPzcyM5FYw4uK-5+DIR4gSYbRnmBZz3oBPiwTIr4HI8+hsMikrGG+N6jAA+jYOGQDhwiFGQBFuGR8BgDh3A4dgeEsHhvCoUivZ0G+yIKHhdGEcRRRkdgKDgpARZ4i0FFwFRNHcYxAAed5pjoxDvs2il6DoxgAPZ9HEAAO2kADoAM6GX0AC2vAabIEyzlQdjmpkhC1EQ2aGKU5YZk4PyQBAMi2BMvBAA) is a circuit with a docked scope at the bottom and with two undocked scopes I've deliberately nicely placed above their respective input source.  Here is a *screenshot* from my browser:

[image]

However, when I "export as image" (which is PNG), I get:

[image]

and when I "export as SVG", I get:

[image]

There are two bugs:

(1) Both the PNG and SVG export failed to position the undocked scopes in the same position as they were laid out in my browser. (The PNG's scope gets shifted vertically up, and the SVG's scope instead gets shifted a little down and left...I haven't investigated further what conditions cause it to shift in a particular manner.)

(2) Also both the PNG and SVG export don't include the docked scope.  (While it could be said that's a feature not a bug, I consider the more correct behavior would be to include the docked scope...presumably the user setup the scope because the consider it important and so they probably would want the scope included in the export, and if they didn't want the scope in the export then they could simply delete the scope before they press export).

ericfont, 2023-06-10:
For the record I am using version 2.8.0js and am on Chrome Version 114.0.5735.110 (Official Build) (64-bit).

pfalstad, 2024-10-24:
fixed

closed 2024-10-24 by pfalstad