sharpie7/circuitjs1 #881: Greater range for Horizontal scale in scope properties

Kain6622, 2023-07-29:
Would it be possible to add additional horizontal scale beyond 200us/div for high frequency circuits, currently can't view the frequency beyond ~10khz. Thanks

pfalstad, 2023-07-29:
If you decrease the time step (Options->Other Options) you can decrease the scale.

Kain6622, 2023-07-29:
Thank you for your response, I tried as you asked and it worked up to about 40khz with time step size at 1n and horizontal scale of the scope set to 1us/div but if I try to increase the frequency past 40khz it glitches out in the scope (goes from showing a measurable frequency to as though I'm pumping 100MHz into it).

I fiddled around with the circuit (it's a variable 555 square wave generator) and have noticed that if I change the value of the variable resistor (lower the range) in the RC part of the circuit it allows me to see higher frequency to a point before the same glitch happens, it seems that the smaller I make the variables resistor range the high the scope can measure to a point before it glitches out.  I've attached the circuit file for inspection.
[cleanercircuit (1).txt](https://github.com/sharpie7/circuitjs1/files/12207953/cleanercircuit.1.txt)

pfalstad, 2023-08-05:
I don't think it's a glitch.  I think the frequency goes up exponentially as you adjust the potentiometer.  I replaced it with two resistors so I could adjust the resistance more carefully, and it seems to become more and more sensitive to changes as you get close to 32.9k/17.1k.

closed 2024-11-29 by pfalstad