sharpie7/circuitjs1 #882: Project has poor drawing performance

Vadimatorik, 2023-07-29:
I am doing a big project. The project contains many nested schemas. When I want to draw a line, after I release the left mouse button, the application freezes for 1-10 seconds. Simulation is disabled. I think the application is trying to draw all components from subcircuits. In total, the project has 10-20 thousand elements (but I plan 200 thousand).

What can you suggest on this issue?

pfalstad, 2023-08-05:
This simulator can't handle a project that large.  Even when simulation is disabled, we analyze the circuit when something changes and that gets very slow as the circuit gets large.

Noitarud, 2023-10-09:
Can an option to suspend analysis be added to the Edit menu? Highlight mistakes (highlight individually or tell user one at a time) after resume attempt, i may have fixed some by then.

Noitarud, 2023-10-09:
Maybe we can "network" multiple simulators (run separate instances on multiple CPUs), the subcircuits will have uplinking inputs and outputs that conduct to whichever other instance, should need a server thread.

Noitarud, 2023-10-09:
@Vadimatorik A cunning way to do this is to draw it up in Microsoft PowerPoint then make a macro to observe every object on the slide and convert the coordinates to the script compatible for use here. I had already thought of this earlier for the purpose of having an object drawing program (shapes, etc) that can work between platforms using VB (even do animations, even in PPT97 with peterson's timerobj).

Post samples of your object-getter and conversion VB code 😎.