sharpie7/circuitjs1 #883: Feature: Add Separate time stamps to different scopes.

Kain6622, 2023-08-02:
Would it be possible to add the feature to allow each instance of a scope to have it's own adjustable time stamp, in the instance of having a very fast frequency and a slower one we currently need to choose which we want to view at the sacrifice to the other when the frequencies differ a lot.

Thank you

pfalstad, 2023-08-04:
you mean the timestep?  the timestep has to be global.  But it's very easy to just increase the range of the horizontal scale slider, so you can have one scope much slower than the other one.

Kain6622, 2023-08-05:
Yes I meant timestep, sorry.  I have tried adjusting the horizontal scale but it is limited to a range based on the timestep, if I have a timestep of 10p I get the horizontal scale range of 500ns/div - 200ps/div (this allows me to view one scope fine) but on one of my scopes I require a horizontal scale in the range of 5ms/div - 2us/div but can't get that unless I set the timestep to 100n (which then messes up the first scope due to scale.  Any chance of allowing a custom range for the horizontal scale or would that take a big performance hit due to all the calculations.

pfalstad, 2023-08-05:
That is a big range.  I could make the range wider but the second scope would be extremely slow.