sharpie7/circuitjs1 #888: violation of Kirchhoff's law with crystals

Westfuchs, 2023-08-16:
When playing around with crystals, I realized that current vanishes or appears from the wires in the connection nodes of the crystal.
Or maybe the current is displayed the wrong way?

Example circuit:
https://tinyurl.com/24fabks2

pfalstad, 2023-08-16:
Hmm, weird.  I think that what's happening is that the current calculation is lagging behind one timestep.  This normally isn't a big deal, but here, the timestep is too large so the current is switching direction every timestep.  You don't notice because we are running an even number of timesteps every frame, but you can see it if you view the current in the scope.

Here is a version with a smaller timestep:

https://tinyurl.com/24oy5wdb

It looks much better but it's still lagging.  I will fix that lag though.

esaruoho, 2026-03-03:
Fixed in v3-dev — crystal current display fixed (Kirchhoff violation resolved).