sharpie7/circuitjs1 #892: offlineversion in winxp sp3 , cannot start "AquireSRWLockExclusive" is missing in KERNEL32.dll

carypt, 2023-09-07:
in windows xp 32bit sp3 when i want to run the offlineversion of circuitjs1-win : "circuitjs1.exe" i get the reply-message (in german):  "Der Prozedureinsprungpunkt "AquireSRWLockExclusive" wurde in der DLL "KERNEL32.dll" nicht gefunden." that means : AcquireSRWlockExclusive" not found in KERNEL32.dll. so i cannot run the nice program on my toughbook. well , i made it possible to run google chrome on that device, so i can use the browser edition of circuitjs1, but i would like to use offline. 
(here is links to annother discussion: https://github.com/rclone/rclone/issues/1481, and from microsoft about :https://learn.microsoft.com/en-us/windows/win32/api/synchapi/nf-synchapi-releasesrwlockexclusive) but that did not help me any.

can i use a different KERNEL.dll from a higher windows (7?) instead by just replacing into winxp ?

can i get an older version of circuitjs ? is it possible to get a java-version ? or can i get the a html-version to start it in offline situation. (the java zip archive seems to be very small in relation to the windows offline version : http://www.falstad.com/circuit-java/ )

its a nice, easy to use, educating program that i like to use in future. its sad i cannot use on older availiable hardware.
ok, thanks , anyway

pfalstad, 2023-09-07:
The circuitjs executable is just electron.  It's a program that basically just packages javascript/html and runs it locally using a browser.  You could look for a version of electron that runs on XP.

Or maybe you could run a web server on your PC, and serve the contents of the zip archive on it.  The file you want to load is circuitjs1/resources/app/war/index.html .

Using a file: URL wouldn't work because of browser security restrictions, but maybe those could be disabled?

carypt, 2023-09-07:
ok, thank you for your reply. as i understand from your message, the windows-offline-version of circuitjs1 (circuitjs1-win.zip) is an java-electron-environment that runs the offline content of circuitsjs1, that use somehow html5-features, and requires a browser-like system. (lushprojects says that it needs a browser (https://lushprojects.com/circuitjs/ "browser and computer requirements") .

that makes it much clearer now, and yes the offline-content : circuitjs1/resources/app/war/index.html  runs well right in the offline browser (chrome) , also does the java-zip run in java itself. ok , then i will try to make it run in browser or electron(xp-version) in win-xp.  

ok, fine, problem solved. thank you for make me understand.

Noitarud, 2023-11-26:
Is it possible to get the Java version? The link is there…
https://www.falstad.com/circuit-java/circuit.zip