sharpie7/circuitjs1 #897: Separate Ground References

agesuale, 2023-10-23:
Hello,
I cannot seem to find a way to have two separate grounds (battery and chassis) in a circuit.  You can see that when I connect a circuit referenced to one ground to a separate reference ground current flows when it should not because they should be a separate reference. Is there a way to assign net labels to the grounds so that they are separated?
[image]

pfalstad, 2023-10-24:
The ground symbol is always at 0 volts.  I'm not sure what voltage you want the different grounds to be at.

If you want CHASSIS_GND to be at a different voltage, then you could just use a label CHASSIS_GND instead of the ground symbol.

If you want to post an example circuit, please use File->Export As Link instead of a screenshot so I can try the circuit.

agesuale, 2023-10-24:
Hi Paul,
Thanks for getting back to me. Just labeling chassis_gnd instead of a ground symbol works for me!

It may be useful in the future to allow for GNDs to be considered and named as separate nets.  Consider the case where you have two batteries in a system that are isolated from each other with separate ground reference. (a circuit very common in HV systems where isolation from the HV side is important).  Forcing the GND symbol to always be a global 0 volts can be problematic since you have to use a node label instead of a GND symbol on one side of your circuit. Hope this makes sense!