sharpie7/circuitjs1 #901: Feature-highlight neighbours

Noitarud, 2023-11-01:
Sometimes I mesh my components together and can hardly tell the immediate circuit apart, can the neighbouring wires and components get a little highlight when I hover over something?

pfalstad, 2023-11-01:
I'm not sure what you're asking here.  You want neighboring components to highlighted when you hover over something?  How would you be able to tell which component is selected then?

Noitarud, 2023-11-02:
I was envisioning the neighbouring components to be blue (rather than cyan), I did not specify a colour just in case you immediately thought of something/someway more comfortable.
Adding blue (in a running circuit) will not work because (positive)green+blue would look cyan, unless you want to also change the highlight colour.