sharpie7/circuitjs1 #911: Bug-dialogue missing for logic and switches

Noitarud, 2023-11-16:
In the older version of circuit you could rightclick/edit on (s)witches and (L)ogic inputs to change them to momentary, here if i avoid the switching area and double tap the body nothing happens (not even toggle as in the older). I was expecting a dialogue to come up to adjust •Momentary •high voltage •low voltage, the fact it doesn't toggle on body taps could mean a dialogue was intended but not finished.

_As for symbol adjustment proposals:
•Maybe you could "pad" the switch section with a thicker line to indicate a larger area you could tap on to either side of the key.
•logic input and (M)logic outputs appear identical, perhaps you could circle the logic inputs to make them consistent with other voltage sources._

pfalstad, 2023-11-16:
yes this still works.  Are you on a mobile device?  Use long press.

Noitarud, 2023-11-17:
I am and I have just figured it out, I am using duck duck go and even if the applet is in fullscreen the browser can scroll (to hide the browser's header) which means I am moving ever so slightly, so I need some discipline during the long press to avoid a drag (or draw).
Could you make it insensitive to dragging unless a whole grid span has been traversed?