sharpie7/circuitjs1 #912: LED Flashing circuit runs very slow compared to real breadboard

jpitz31, 2023-11-21:
Because I have two loops of LED's I get a wire loop detected error. According to what I read, the solution is to add a very small resistor to each loop.  When I do, regardless of the simulation speed or the current speed settings, the LED flash is very slow in changing states.

Enclosed is a copy of the circuit file. 

The source is 9V DC.. The chip is a 555.  The LEDs are wired in an alternating pattern, positive and negative.
I added an .txt extension to the Falstad file to GitHub would accept the file upload.

Thanks
[Christmas LED.txt](https://github.com/sharpie7/circuitjs1/files/13420979/Christmas.LED.txt)

Joe

mpictor, 2023-11-24:
Are you saying the led blinking on your screen isn't as fast as one on a breadboard, or that the simulation's predicted timing is different from what you observe with a breadboard? These are different things :slightly_smiling_face: 

If the former, this is to be expected; the simulation takes a lot of computation. Changing the time step (Options | Other Options) can change this, but of course it's a tradeoff between speed and accuracy.

If the latter, how quickly are you expecting it to change? When I simulate with 33k, 10uF, and using the output pin rather than the discharge pin, I get ~2.2 Hz - see screenshot below.

Since the LEDs are only attached to the timer's output, the number present won't affect how fast the timer in the simulation cycles. They can still slow the simulator itself down, so I'd put no more than two in the circuit while figuring out timing. If you're unlucky, the extra components can also cause the simulator to act up.

Having a pair of 1k resistors in series across the battery will drain it much faster than the circuit would otherwise. Instead, I'd use 3 wires between the LED array and the rest of the circuit - one each from B+, B-, and the timer output. The timer output will go to all LEDs as it does currently, while the other two each go through a resistor and then to alternating LEDs.

[circuit link](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcBmBsBsyCcyBMAWLSXXSAdhAQiQUkoFMBaMMAKDHVxH3wA4RUXPnUEguAE3oAzAIYBXADYAXRgvrjwULTEhsOQ3nXwdufbiYnT5y1eq0Q6Ovehpl03QyHSR+wymFx3HXJkdCwsMAR8KgRcMD50VgB3EDJIfEo3VPR+BCzIZOzcgKLMoML0fCMzb18zAtIsbkrKHkpcXLbHWHhIcP6BwfDuaGQ+nDD0eOR43gpgiDYUvPcojNjOjIKU-Ba15oyefm2D035as6gKsi4VrxuvHyuU9AeLwzoLk4-H-h-Rb5wASxDwiEEFABOAl46zayBhXgI9nghXhGU46yyGOeZURmNWXVYUN2sMOLS+AjGcEKJN+9y4X1YADdcS53Gl0QggloqiAmo4oNAEIU7mzWUiThzcVKxSc0a0MvLjOUUkqTEdzCrQcCDEYTJL0qlDchQkatkTKezjabaXQwCjVTaWib3ADUabRBq3QB5Okai7K7SsX0-d54OmOQoyrlm2MFAD2AjouBjIhyZAy9sFDjEYn5rCAA) and screenshot
[image] and

jpitz31, 2023-11-26:
Thanks for the quick feedback.  I will try your suggestions. 

Joe