sharpie7/circuitjs1 #915: Please describe how to operate Plot XY - or is this it?

Noitarud, 2023-11-26:
Okay,

- if you download the. Jar zip file, the Y input is a little hidden, if there is a preceeding scope on a scope probe (probably last drawn), that is used. Else, when select Y is pressed it will choose the last drawn scope probe. Perhaps if you choose 'select Y' it should automatically select to draw scope probes.

- The new one in the browser, you create a test point then add it to scope, scope probe doesn't appear to work.

Are there other ways?

pfalstad, 2023-11-27:
For java, you use "Select Y" multiple times to select the scope probe you want to use for Y.

For the browser, combine two plots in the same scope, one for X, one for Y (using right mouse -> combine) then select "Plot X/Y"

Noitarud, 2023-12-13:
Thanks; it needed to be said.
Maybe the Java could have "reselect Y probe" (which implies probes necessary).

Maybe you could put a '"help" GitHub label on this issue.