sharpie7/circuitjs1 #921: Scope Properties toolbar hides scope on resizing in XY Mode

ArmoredRodent, 2024-01-01:
If you resize scope window with scope in X-Y Plots mode, what appears to be the Scope Properties toolbar appears and hides the lower portion of scope. 

Sample file to illustrate the problem:

$ 3 0.000005 11.558428452718767 57 5 50 8
R 400 96 400 64 0 0 40 -15 0 0 0.5
w 400 96 400 176 1
g 400 280 400 296 0 0
p 336 208 368 232 0 0 0
R 336 192 288 192 0 3 100 5 -4 0 0.5
r 400 232 400 280 0 5
w 816 192 848 192 0
368 816 192 816 152 0 0
164 624 144 640 144 2049 4 0 5 0 0 false 13
166 672 144 688 144 2049 4
g 616 144 616 160 0 0
w 624 144 616 144 0
R 624 192 568 192 0 2 100 2.5 2.5 3.141592653589793 0.5
R 720 144 760 144 0 0 40 -7.5 0 0 0.5
w 720 192 752 192 0
r 912 120 912 176 0 100
g 912 232 912 248 0 0
R 912 80 912 48 0 3 100 20 -20 0 0.5
w 400 208 400 232 0
w 336 192 336 208 0
w 400 232 368 232 0
x 322 235 332 238 4 16 V
x 332 240 344 243 4 12 gs
x 956 205 968 208 4 12 ds
x 947 196 957 199 4 16 V
x 384 149 390 152 4 12 d
x 378 142 382 145 4 16 I
x 892 106 896 109 4 16 I
x 898 113 904 116 4 12 d
w 912 80 912 120 1
p 936 176 936 208 0 0 0
w 936 208 912 208 0
w 936 176 912 176 0
w 912 208 912 232 0
212 752 192 768 192 1 2 1*(a-b)-1.5
w 752 208 744 208 0
g 744 208 744 224 0 0
w 800 208 808 208 0
g 808 208 808 224 0 0
w 816 192 800 192 0
f 336 192 400 192 1 1.54 0.02
f 848 192 912 192 1 1.5 0.02
o 3 64 0 12355 2e-23 1.6e-24 0 2 1 3 EnMOSFET\sVgs\svs\sId
o 41 64 0 12359 40 0.35 1 2 41 3 EnMOSFET\sVds\svs\sId

pfalstad, 2026-05-04:
I'm not seeing that, but I am seeing that it's trying to draw into the lower part of the scope and all I see there is black.

ArmoredRodent, 2026-10-08:
Fixes sharpie7/circuitjs1#921.

Scope.setRect() only resets the graph (which reallocates the 2D plot's off-screen image) on a height change when plot2d.plotXY is set. "Show V vs I" plots also draw through plot2d but have plotXY == false, so when such a scope is made taller the axes are drawn at the new size while the trace image keeps its old height, leaving a blank area at the bottom of the scope.

Checking plot2d.enabled instead covers both 2D modes (V vs I and Plot X/Y). One-line change; width changes were already handled for all modes.

Specifically:
`// Scope.setRect(), line 321`
`if (this.rect.width != w || (plot2d.enabled && this.rect.height != h))`

found and fixed with Claude