sharpie7/circuitjs1 #922: Offline macOS Arm version is corrupted

OneCodeCZ, 2024-01-08:
There is only one file "Elektron" in DMG. After opening it Mac says: “Electron” is damaged and can’t be opened. You should eject the disk image. Intel version works just fine.

pfalstad, 2024-01-08:
Yes, I didn't build that correctly, apparently. I don't have an ARM mac to test it on. Someone said they fixed it by running this command in terminal:

xattr -cr Electron.app

OneCodeCZ, 2024-01-08:
I did not build it. I got it from your Website: https://www.falstad.com/circuit/offline/CircuitJS1-macarm.dmg

pfalstad, 2024-01-08:
Glad that worked.  I don't think it will affect performance much.  I would think it would have the same performance as using the website with Chrome.

closed 2024-01-08 by pfalstad