sharpie7/circuitjs1 #926: Question: Memory

lukacs-leon, 2024-02-18:
How can I add a ROM or a flash memory to my circuit, or if it is not there, can it be included?

pfalstad, 2024-02-18:
Use the Static Ram under "digital chips".

You can use it like a ROM.  Just don't write to it.  🙂

lukacs-leon, 2024-02-19:
Thanks @pfalstad

lukacs-leon, 2024-02-19:
How can I write data to a static RAM?

lukacs-leon, 2024-02-19:
Under the edit, content?

pfalstad, 2024-02-19:
yes, you can load the static ram using right click, Edit, Contents.

What's your other question?  I can't tell what's on the static RAM by looking at a picture of it.

lukacs-leon, 2024-02-19:
what can i do with which leg?

pfalstad, 2024-02-19:
The WE is "write enable".  That should be low (0V) when writing, high (5V) when reading.  So if you are making a ROM, that should always be 5V.

The OE is "output enable".  The RAM is enabled when low (0V), disabled when high (5V).

A0-A3 are the address pins, and D0-D3 are the data pins.

lukacs-leon, 2024-02-19:
how big is this memory? how many Bytes / Kb?

lukacs-leon, 2024-02-19:
does it have four serial ports or can I send four bits on a parallel port?

pfalstad, 2024-02-19:
well, it's got 4 address pins and 4 data pins so it can store 2^4 = 16 4-bit words, or 8 bytes.

I guess you could call it a parallel port.  The data bits are accessed in parallel.

lukacs-leon, 2024-02-19:
could you give me an example of something when I enter something in advance on the edit and then it reads it and writes it to a scope with an out?