sharpie7/circuitjs1 #930: (Measurement) tools only work when connected directly with the junction points of an element

msedi, 2024-03-14:
First: Awesome tool!!!

It took me a while to understand that I cannot simply add a voltmeter directly on some connection wire, but instead needed to add the voltmeter to a junction point of e.g. a resistor (see the attached image for an example).

You can see in the image that it is necessary to connect the voltmeter with the junction point of the resistor. Everything else is considered an invalid connection. Is this intended. Would it make sense to allow every point on the wire but internally snap it to some junction point. There might be more junction points available, but in the end it would matter which one is chosen.

[image]

ArmoredRodent, 2024-03-17:
Common problem for new users. 

Place your mouse close to Red node, highlighting trace umderneath. [Ctrl] + LMB will add a junction to trace and Red node will disappear.

msedi, 2024-03-18:
Perfect. Thanks for the hint.

closed 2024-03-18 by msedi