sharpie7/circuitjs1 #940: Single pulse generator

remfan77, 2024-05-07:
Hello, I would like to use a **single** pulse voltage generator.
I only need as parameter : 
- pulse length
- pulse init time

The current pulse object requires to enter a frequency.

Maybe it's already possible and I'm not able to use it  ?

Thanks for support and for this useful tool.

Best regards,
Paolo

pfalstad, 2024-05-07:
Here is one that starts at time 0.1 and ends at time 0.2.  You can change the times using right click, edit, then change the output function.

https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWEBmB0BskBMAWdn0B2SQwgTggUhCSpoFMBaMMAKCzCxCK+XQA4Q-OCgFRxAZwAu9AA4AKKY2hgAlACppc+dCyMpGhKwDmIQmJwjzgnGHTjIrAE5mxfQf3Q5Rg6mHgmQl4+Qd6W1BGsQA

remfan77, 2024-05-07:
Thanks for the example.
Best regards