sharpie7/circuitjs1 #949: 1N34 (germanium diode) incorrectly simulated?

ericfont, 2024-06-07:
Try [this simulation of a 1N34](https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWEAmAHA6BOAzJbAWAdm2QQUMnxAUhWtoQFMBaMMAKACUVVURn8fNH2aYAbFBDZwtGKSjR5s6AnYATHiMGapyOmAByBdgHMdJHWKqz2Ae3AgrksGkh6Y+TF+8+vY8dawcBC0ehDYdg5OtC6okCFB8QrwIfpSkRDRIPhgYtJgijTwxSX4LGCU1NAuJbVu5XgOYVIgAJIAOgDOAG6dXQBq7EA). When inputting voltage from 0 to 500 mV, the current only reaches 1.363 ***micro*** amps:

[image]

But germanium diodes are typically said to have a rough threshold around 300 mV all datasheets and plots of 1N34 diodes I can find on the internet show the current should be well into the ***milli*** amp range by the time its voltage is 500 mV, [for instance:](https://w140.com/tekwiki/images/1/1d/Sylvania_Germanium_Diodes.pdf)

[image]

I looked at https://github.com/sharpie7/circuitjs1/blob/dd47cf43a1baa91e2d6ae0f5884ccf0f589989bf/src/com/lushprojects/circuitjs1/client/DiodeModel.java#L90 but it seems to match what  https://www.allaboutcircuits.com/textbook/semiconductors/chpt-3/spice-models gives. Though there are some parameters left out from that...maybe they are necessary?  Or maybe there is some other error simulating it in DiodeModel.java?

Thanks as always for such a great tool.

ericfont, 2024-06-07:
I don't notice any glaring issues with the two Shottkey diodes, which are also low-threshold diodes and match my limited understanding. But the 1N34 seems way off...maybe it should just be deleted from that menu if fixing its simulation is too hard.

pfalstad, 2024-06-07:
Have you seen this:

https://forum.allaboutcircuits.com/threads/error-in-1n34-spice-model.48758/

I tried it with MacSpice and got exactly the same results.  I guess the model is just bad.  Deleting it from the menu seems like a good idea.

ericfont, 2024-06-07:
Ahh.  Yeah...just delete it for now.