sharpie7/circuitjs1 #957: Resistance can affect the signal phase

nvnv-sunny, 2024-08-08:
A resistor in series with an AC source, when the resistor's terminals are adjusted, can cause the signal  to be inverted by 180 degrees. This may lead to confusion for users when simulating related circuits.

pfalstad, 2024-08-08:
How did you create the resistors?  It matters if you drag left to right, or right to left.  the positive terminal should be placed first, and then drag to the negative terminal.  If you do it wrong, you can use right click then "Swap Terminals"

nvnv-sunny, 2024-08-08:
I understand what you mean, but this phenomenon does not exist in real circuits. I don't fully understand the operating principles of this system, but I suggest that it might be helpful to indicate the direction of current on the component graphics,Includes other components.This way, users will know which direction of current the oscilloscope is capturing signals from, avoiding the current situation that leads to erroneous simulation results, especially for beginners who are just getting into circuits

ArmoredRodent, 2024-12-31:
The phenomenon exsists in real circuits.  Put your scope probes on opposite sides and you will get your waveforms.

You are putting a component in and putting it in the scope, so how you create the component must have a bearing on how it will be shown in the scope.

If you include the text, someone can verify the issue.

pfalstad, 2024-12-31:
Agree that adding a +/- indicator on component terminals when displayed in the scope would be helpful.