sharpie7/circuitjs1 #959: Convergence failed!

badrussians, 2024-08-25:
If switch a single switch and achieve currents in picoamps,
then it consistently gives an error on a simple circuit.
[image]
[circuit с 1 mosfet PNP.txt](https://github.com/user-attachments/files/16739828/circuit.1.mosfet.PNP.txt)

badrussians, 2024-08-25:
Functionality requests:
1. And it is possible to make it so that there are, for example, five current speeds - the current should always flow (both small and large)!
2. And make the simulation speed controlled by a number - make a slider from minimum to maximum. Otherwise, it is generally impossible to measure some processes.
3. And make it so that it is possible to reverse the simulation in time, to see what just happened.

ArmoredRodent, 2024-12-31:
Spread your circuit out.  Give the simulation a chance.

[circuit с 1 mosfet PNP_tweaked.txt](https://github.com/user-attachments/files/18282324/circuit.1.mosfet.PNP_tweaked.txt)

badrussians, 2026-01-04:
Unclear answers.
Problem not resolved.

closed 2026-01-04 by badrussians